javaplus/PicoProjects #26: more game ideas

barthba, 2022-12-01:
- fill the LED
- arm moves but button stops it for X time
- percentage of light
- speed up the rotation as you hit it/// rounds each time you have a successful hit?
- rainbow lights after winning or something
- digit displays to show a score or percent or something?
-

barthba, 2022-12-06:
if buzzers are in the plan then it could do a little hooray or sad buzzer upon a hit or miss

javaplus, 2022-12-06:
I love your "another take"... I think having it start slow and then pressing the button to "fire" the laser could be cool and then having a concept of "lives" indicated by lights could be cool...
And if we really wanted to, we could have the button turn the laser on.  It would mean running the positive line of the laser to the pico and not just the power rail so we can use a GPIO pin to turn the laser on and off.  If we did that, then the button press would turn the laser on for a short amount of time and also start the recording of light intensity on the photoresistor and after the laser went off record whether they lost a life or not.

barthba, 2022-12-31:
game has progressed - closing this issue

closed 2022-12-31 by barthba