javaplus/PicoProjects #29: Bad link in README.md

kunderj, 2022-12-28:
The link to "Lab 1" in the main readme gives a [404](https://github.com/javaplus/PicoProjects/blob/main/labs/first_circuit.md). I don't feel like fork/PR-ing just for this. Just say'n.

javaplus, 2022-12-31:
LOL... Thanks... this is a work in progress... all the labs should be linked together in the next week or so and the main Readme will be updated to point to all the labs!

javaplus, 2022-12-31:
Decided to go ahead and fix it.  Updated Readme with links to all the completed labs.  Still have a few final instructions/labs to build a game, but they aren't done yet.  Also, hope to include a parts list in the near future.

@kunderj Thanks for looking at our repo.  Feel free to raise more issues or reach out to me if you have any questions on this material btarlton@gmail.com

PR #49 (2022-12-31): Update Readme with links to all completed labs 15 of them
closes #29
closed 2022-12-31 by javaplus