javaplus/PicoProjects #30: GPIO pin confusion

mbeale0, 2022-12-30:
Forget which lab I realized this in, I might have even commented on it already but it might be helpful to mention that things like "GPIO 15" isn't necessarily the 15th connection point, since it is row 20(specifically thinking of Lab12 but meant to mention this in a previous lab)

barthba, 2022-12-31:
@javaplus when do you plan on explaining headers?  is that earlier in the presentation?  I know you plan on having the handouts but wasn't sure if we should slip a section in on lab 1

javaplus, 2022-12-31:
I'm going to include a picture that lays out the headers pins... was going to be a physical picture but due to cost and time, I'm going to just have a soft copy here in the repo... .. but I think with every lab we have diagrams for wiring, but I'll try to make a note about the pico pins... that is that the physical numbering doesn't match the GPIO pin numbers.

javaplus, 2022-12-31:
Added a new page called /reference/pico_gpio.md to talk about logical numbering of Pico pins versus GP numbering.  Linked to it from the first lab.  May need to link to it from other labs.

PR #50 (2022-12-31): Update 07_blink_led.md
closes #30
PR #51 (2022-12-31): Create pico_gpio.md
closes #30
closed 2022-12-31 by javaplus