javaplus/PicoProjects #34: Lab9 - Confusing problem

mbeale0, 2022-12-30:
- For me trying to think of how to fix the debouncing without the code at first felt a little harder. Looking at the code,
	  it all makes sense but think there was just several little things in there. One thought would be suggesting a time to wait in between
	  switching in the text. In my opinion, telling them to attempt on their own "Fix bouncing by keeping track of timing in between
	  clicks. 500ms is a good time to try" just gives a little grounding point/orientation. Might have just been me though. I am tired =)

barthba, 2022-12-31:
I don't see an issue personally but I'll let @javaplus see if he thinks something should change.

javaplus, 2022-12-31:
Updated the lab to just add another sentence clarifying the approach to take for the solution.

PR #52 (2022-12-31): Update 09_button_debounce.md
Added clarification on how to attempt it yourself.

closes #34
closed 2022-12-31 by javaplus